esp-rs/esp-idf #427: Uart rx buffer length error

Volkalex28, 2024-05-30:
I know that similar problems have already existed, but I did not find a solution there.

I'm trying to configure UartTxDriver and get the error:
"uart: uart_driver_install(1588): uart rx buffer length error"

I don't change the config

let config = Config { baudrate: Hertz(19200), parity: Parity::ParityOdd, ..Default::default() };

Volkalex28, 2024-05-30:
version: 0.43.1
chip: ESP32S3

Volkalex28, 2024-05-30:
It looks like the problem is this

In esp-idf we have [assert](https://github.com/espressif/esp-idf/blob/a322e6bdad4b6675d4597fb2722eea2851ba88cb/components/driver/uart/uart.c#L1588C26-L1588C41)
```
ESP_RETURN_ON_FALSE((rx_buffer_size > UART_HW_FIFO_LEN(uart_num)), ESP_FAIL, UART_TAG, "uart rx buffer length error");
```

but in esp-idf-hal we have [this ](https://github.com/esp-rs/esp-idf-hal/blob/04ad4299a895761d2866dc26859fbde3692ab103/src/uart.rs#L1897) uart_driver_install call
```
esp!(unsafe {
    uart_driver_install(
        UART::port(),
        if rx.is_some() {
            config.rx_fifo_size as _
        } else {
            0
        },
        if tx.is_some() {
            config.tx_fifo_size as _
        } else {
            0
        },
        config.queue_size as _,
        queue.map(|q| q as *mut _).unwrap_or(ptr::null_mut()),
        InterruptType::to_native(config.intr_flags) as i32,
    )
})?;
```

It turns out that if don’t have an rx pin, then we are trying to set the buffer length to 0, but esp-idf does not allow us to do this

ivmarkov, 2024-06-07:
@Volkalex28 Would you like to contribute a PR that addresses this issue?

Volkalex28, 2024-06-09:
@ivmarkov In general, yes
I also have a change for ResetReason. In from impl, remove the panic and replace it with ResetReason::Unknown. Do you want to discuss this elsewhere?

ivmarkov, 2024-06-09:
Let's have two separate PRs for these if you don't mind.

If you want to discuss or in general see what is going on - join the Matrix room which is linked in the readme.

ivmarkov, 2024-06-13:
@Volkalex28 what happened? Are you still interested in contributing a PR?

Volkalex28, 2024-06-13:
@ivmarkov Sorry for delay. I didn't have time to do this at all. I think I'll do it tomorrow!

ivmarkov, 2024-06-13:
No rush. It is just that we are prepping for a release these days, so if you want the fixes to go in, you might need to hurry up.

But we can always do a patch-level release later.