esp-rs/esp-idf #428: Zigbee Support

Nassiel, 2024-05-31:
Hi,

By using `IDF` directly to configure the _Zigbee_ protocol, I was able to connect a device to a _Zigbee_ network and manage it using **Home Assistant**. There are plenty of examples for C there.

Is there any plan to include the same capabilities vía `idf-hal` and add also examples for **HA**?

Thank you!

ivmarkov, 2024-06-07:
@Nassiel You could be the one that drives that effort! 
An initial step would be to PR against `esp-idf-sys`'s `bindings.h` header all the Zigbee headers we need to include from the ESP IDF. This way, all C examples can be (unsafely) re-implemented in Rust by calling into the ESP IDF unsafe Zigbee APIs that would now be available from Rust

Once we have this, we can think of a nice type-safe wrapper for Zigbee in `esp-idf-svc`. But that's only a second step.

The `esp-idf-*` projects are community-driven, so unless someone steps in and contributes a PR, certain functionalities might wait until one of the current maintainers (as in me and Frank) has either free time or use-case to implement this functionality ourselves.

Nassiel, 2024-06-07:
Hi @ivmarkov, I'd love to do it, but I'm struggling to map functions like `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);` for instance, inside the IDF mapping. Is there any documentation of where they are mapped to Rust, naming, data structure, etc.?

ivmarkov, 2024-06-07:
Look at the `esp-idf-sys` crate that I mentioned. It generates the "mappings" (bindings) automatically for you, as long as you `#include` the relevant ESP IDF headers in the `bindings.h` file which lives inside it.

Nassiel, 2024-06-07:
Thanks @ivmarkov let see each other on the Matrix chat :D

closed 2024-06-07 by Nassiel

wouter173, 2024-08-21:
Hello, wondering if any progress was made on the zigbee side of ieee802. 

Just started playing around with esp-rs and would love to move over some of my code from c to rust!