esp-rs/esp-idf #458: [Description fix] spi.rs DMA support description

xiaguangbo, 2024-07-15:
DMA 'N' need change to 'Y'?

```
//! Look at the following table to determine which driver best suits your requirements:
//!
//! |   |                  | SpiDeviceDriver::new | SpiDeviceDriver::new (no CS) | SpiSoftCsDeviceDriver::new | SpiBusDriver::new |
//! |---|------------------|----------------------|------------------------------|----------------------------|-------------------|
//! |   | Managed CS       |       Hardware       |              N               |     Software triggered     |         N         |
//! |   | 1 device         |           Y          |              Y               |              Y             |         Y         |
//! |   | 1-3 devices      |           Y          |              N               |              Y             |         N         |
//! |   | 4-6 devices      |    Only on esp32CX   |              N               |              Y             |         N         |
//! |   | More than 6      |           N          |              N               |              Y             |         N         |
//! |   | DMA              |           N          |              N               |              N             |         N         |
//! |   | Polling transmit |           Y          |              Y               |              Y             |         Y         |
//! |   | ISR transmit     |           Y          |              Y               |              Y             |         Y         |
//! |   | Async support*   |           Y          |              Y               |              Y             |         Y         |
//!
//! * True non-blocking async possible only when all devices attached to the SPI bus are used in async mode (i.e. calling methods `xxx_async()`

```

Vollbrecht, 2024-07-18:
yes that is wrong. The spi driver internally handles all the dma stuff it the user sets to use one of the dma channels. Do you want to update it?

xiaguangbo, 2024-07-18:
> yes that is wrong. The spi driver internally handles all the dma stuff it the user sets to use one of the dma channels. Do you want to update it?

Yes.

xiaguangbo, 2024-07-18:
@Vollbrecht  https://github.com/esp-rs/esp-idf-hal/pull/462

closed 2024-07-18 by Vollbrecht