esp-rs/esp-idf #465: Cannot open .dll

MiniMinerX, 2024-07-23:
I have followed all the installation instructions, making sure my enviornment variables are set and correct. I cannot build because the .dll file is not being opened and I do not know why. I've treid working at this for hours now with not much online to help.  Here is my tomls: [package]
name = "espt12"
version = "0.1.0"
authors = ["MiniMinerX <jessehrupe@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.79"

[[bin]]
name = "espt12"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.49", default-features = false }

[build-dependencies]
embuild = "0.32.0" and then this toml: [toolchain]
channel = "esp" and then finally: [build]
target = "xtensa-esp32-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor" # Select this runner for espflash v3.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32"
RUST_BACKTRACE = "full"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.2.2"

# Workaround for https://github.com/esp-rs/esp-idf-template/issues/174 
CRATE_CC_NO_DEFAULTS = "1"

I ran these commands in the terminal to source the paths: > $Env:LIBCLANG_PATH = "C:\Users\minim\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\bin\libclang.dll"
>> $Env:PATH = "C:\Users\minim\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\bin;" + $Env:PATH
>> $Env:PATH = "C:\Users\minim\.rustup\toolchains\esp\xtensa-esp-elf\bin;" + $Env:PATH
PS C:\espt12>

Here is my output after doing cargo build --release

Note the line: Unable to find libclang: "the `libclang` shared library at C:\\Users\\minim\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"

 PS C:\espt12> cargo build --release
   Compiling bindgen v0.69.4
   Compiling embuild v0.32.0
   Compiling esp-idf-sys v0.35.0
   Compiling esp-idf-hal v0.44.1
   Compiling esp-idf-svc v0.49.1
[... 216 lines cut ...]
    13:     0x7ff6834bf604 - core::result::Result<T,E>::expect::hb3309c3a1c3c3291
    14:     0x7ff6834752bc - core::ops::function::FnOnce::call_once::h5d7cdaa71849ee6d
    15:     0x7ff6834f7c3b - lazy_static::lazy::Lazy<T>::get::{{closure}}::h703506b50d675ab2
    16:     0x7ff683517c40 - std::sync::once::Once::call_once::{{closure}}::hb294324ebc6a8bd5
    17:     0x7ff683adb3b4 - std::sys::sync::once::queue::Once::call::h6b6b7976e380e78c
    18:     0x7ff683517ae0 - std::sync::once::Once::call_once::hc09c69e5243b26ad
    19:     0x7ff683515ff4 - <bindgen::ensure_libclang_is_loaded::LIBCLANG as core::ops::deref::Deref>::deref::h7a23bd74ad345d54
    20:     0x7ff68350ece9 - bindgen::deprecated_target_diagnostic::hf7152a9482ceb835
    21:     0x7ff68350fdbe - <bindgen::BindgenError as core::fmt::Display>::fmt::h3e8cdbf0341aa52e
    22:     0x7ff68350e3c9 - bindgen::Builder::generate::h1c3ad78b548d2a94
    23:     0x7ff68318671c - <build_script_build::BindgenCallbacks as bindgen::callbacks::ParseCallbacks>::int_macro::{{closure}}::ha64a0ff38e86e3d0
    24:     0x7ff6831b2696 - core::ops::function::FnOnce::call_once::h6d0f87376d070dbb
    25:     0x7ff6831e5779 - std::sys_common::backtrace::__rust_begin_short_backtrace::he5abf919a1dafae7
    26:     0x7ff6831a5a2c - std::rt::lang_start::{{closure}}::h832a9944a0b32cb9
    27:     0x7ff683a97afe - std::panicking::try::h224dfc5b82599353
    28:     0x7ff683aa7f4e - std::rt::lang_start_internal::h243ea0028ca15fb1
    29:     0x7ff6831a5a07 - std::rt::lang_start::h01c8cabd9d0af738
    30:     0x7ff683188ba9 - main
    31:     0x7ff683ad9650 - invoke_main
                                 at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    32:     0x7ff683ad9650 - __scrt_common_main_seh
                                 at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    33:     0x7ffc39f8257d - BaseThreadInitThunk
    34:     0x7ffc3b1eaf28 - RtlUserThreadStart
PS C:\espt12>,

Vollbrecht, 2024-07-28:
That may be a espup problem. We have similar reports from other uses and they could resolve it by downgrading to espup version 11 and installing the toolchain with that espup version. There is an open issue [here](https://github.com/esp-rs/espup/issues/440)

MiniMinerX, 2024-07-28:
That makes sense. I'll try that today. For now I gave up on rancid windows and am running Ubuntu inside of vmware workstation. The Linux setup of esp32 for me was a breeze, so I've been developing on that. I also tried wsl, but my esp32 needed the ch341 drivers, so I had to recompile the wsl kernel with them enabled. Then I broke wsl usbipd for good it seems, so vmware is my savior.

ivmarkov, 2024-08-07:
> That makes sense. I'll try that today. For now I gave up on rancid windows and am running Ubuntu inside of vmware workstation. The Linux setup of esp32 for me was a breeze, so I've been developing [...]

When I used to work with WSL, I used to do all development from Linux, but then for flashing I just copied the resulting ELF to the Windows partition and then used the `espflash` utility natively compiled for Windows (with `cargo install espflash`). This way I did not have to deal with obscure usb drivers on the Linux level.

ivmarkov, 2024-08-07:
If I remember correctly, I think you can call native windows binaries from the Linux shell, so with some bash scripting you can further automate calling the Windows-compiled `espflash` from WSL.