esp-rs/esp-idf #466: Use SPI Bus for multiple Devices

agmes4, 2024-07-31:
I wanna use the Spi Bus for multiple Spi Devices Problem here is that the config for one device is a bit different.
(It is taking data on raising Edge the other on falling.)

So is there any way to change config on runtime of a SpiDeviceDriver or to create two different one so I can pass one to each devices?

Vollbrecht, 2024-07-31:
Why not just create two or more SpiDeviceDriver instance each with its own config that represents each MODE option?

agmes4, 2024-07-31:
Tried that already but the driver con just be passed once cause then it get moved and after that I can not create a new one

Vollbrecht, 2024-07-31:
A SpiDeviceDriver takes a Borrow<SpiDriver>. If you are not familiar with Borrow you can read about [here](https://doc.rust-lang.org/std/borrow/trait.Borrow.html). It allows you to not only pass by `&SpiDriver` but also everything that implements Borrow like an 'Arc<SpiDriver>` etc.

agmes4, 2024-07-31:
ah thanks refactoring solved it for me the async task messed things a bit up

closed 2024-07-31 by agmes4