esp-rs/esp-idf #475: UartDriver as `embedded_io::Read` blocks until buffer is full

DaneSlattery, 2024-08-21:
In the below code snippet from the uart loopback. 
We know that the uart TX has written 1 byte.
The receive buffer is `[0;10]` , which is typical when we don't know how large the receive buffer is.

In this case, the read blocks forever, instead of popping 1 byte from the UART buffer (into `buf`) and returning `1`. If the buffer is `[0;1]`, the read returns instantly.

This means that the `embedded_io::Read` contract is not being upheld:
```
/// Read some bytes from this source into the specified buffer, returning how many bytes were read.
    ///
    /// If no bytes are currently available to read, this function blocks until at least one byte is available.
    ///
    /// If bytes are available, a non-zero amount of bytes is read to the beginning of `buf`, and the amount
    /// is returned. It is not guaranteed that *all* available bytes are returned, it is possible for the
    /// implementation to read an amount of bytes less than `buf.len()` while there are more bytes immediately
    /// available.
    ///
    /// If the reader is at end-of-file (EOF), `Ok(0)` is returned. There is no guarantee that a reader at EOF
    /// will always be so in the future, for example a reader can stop being at EOF if another process appends
    /// more bytes to the underlying file.
    ///
    /// If `buf.len() == 0`, `read` returns without blocking, with either `Ok(0)` or an error.
    /// The `Ok(0)` doesn't indicate EOF, unlike when called with a non-empty buffer.
```

```rust
//! UART loopback test
//!
//! Folowing pins are used:
//! TX    GPIO12
//! RX    GPIO13
//!
//! Depending on your target and the board you are using you have to change the pins.
//!
//! This example transfers data via UART.
//! Connect TX and RX pins to see the outgoing data is read as incoming data.

use embedded_io::Read;
use esp_idf_hal::delay::BLOCK;
use esp_idf_hal::gpio;
use esp_idf_hal::peripherals::Peripherals;
use esp_idf_hal::prelude::*;
use esp_idf_hal::uart::*;
use esp_idf_sys::EspError;

fn main() -> anyhow::Result<()> {
    esp_idf_hal::sys::link_patches();

    let peripherals = Peripherals::take()?;
    let tx = peripherals.pins.gpio12;
    let rx = peripherals.pins.gpio13;

    println!("Starting UART loopback test");
    let config = config::Config::new().baudrate(Hertz(115_200));
    let mut uart = UartDriver::new(
        peripherals.uart1,
        tx,
        rx,
        Option::<gpio::Gpio0>::None,
        Option::<gpio::Gpio1>::None,
        &config,
    )?;

    loop {
        uart.write(&[0xaa])?;

        read(&mut uart)?;

    }
}

fn read<R: embedded_io::Read>(r: &mut R) -> Result<(), anyhow::Error> {
    let mut buf = [0u8; 10];
    match r.read(&mut buf) {
        Err(x) => println!("Error reading"),
        Ok(x) => println!("Read :{:?} ", &buf[..x]),
    };
    Ok(())
}
```

ivmarkov, 2024-08-22:
You are right! 

I have mis-interpreted what `uart_read_bytes` actually does, and (sorry for pointing the finger at ESP IDF) BUT - yes - the signature of that function is VERY misleading, as it looks exactly like the C IO `read` call, (and like the Rust `Read` trait semantics), yet it behaves VERY differently!!

You can see the gory details [here](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_uart/src/uart.c#L1493C5-L1493C20). I can explain what is weird about this C code but it will be a long explanation, and hopefully you can decypher by yourself what is wrong with it.

In fact, I would say the current behavior of `uart_read_bytes` is not simply unexpected (as it does not follow the semantics of a regular `read` IO syscall), but it is very incovenient for user code trying to utilize `uart_read_bytes` as that user code has to read the input data byte-by-byte... :(

ivmarkov, 2024-08-22:
@DaneSlattery ^^^

DaneSlattery, 2024-08-22:
That fixes it, thank you! Happy with the solution you came to.

closed 2024-08-22 by DaneSlattery