esp-rs/esp-idf #476: Sharing a SpiDriver between multiple Devices with different configs

agmes4, 2024-08-28:
Im currently trying to get multiple SpiDeviceDrivers out of SpiDriver cause the devices have different configs. Borrowing worked for compiling but all the time I try to create the second DeviceDriver the esp get stuck in execution. 

Is there first an easy way to share to share a SpiBus here with out these issues and is there additional documentation for such cases? 
I tryed
```
let mut spi_device_driver =
        SpiBusDriver::new(&mut spi, &config).expect("Failed to init spi flash!");
spi_device_driver.write(b"as");
spi_device_driver.write(b"as");
spi_device_driver.write(b"as");
spi_device_driver.write(b"as");
spi_device_driver.write(b"as");
info!("have send");
let config = config::Config::new()
        .baudrate(4000000.into())
        .bit_order(config::BitOrder::MsbFirst)
        .write_only(true)
        .input_delay_ns(0)
        .data_mode(MODE_0);
    
let mut spi_device_driver2 =
        SpiBusDriver::new(&mut spi, &config).expect("Failed to init spi flash!");
spi_device_driver2.write(b"as");
```

agmes4, 2024-08-28:
the Problem here is that my display needs to have a minor delay between 2 bytes what I do in the device via manually pulling the CS to high and the later again to low. I think I found a way by passing Option::<AnyIOPin>::None as cs it is possible to create without setting a pin  thanks.

Vollbrecht, 2024-08-28:
you can have SpiDeviceDriver A in thread A and SpiDeviceDriver B in thread B yes. Though if you want to share SpiDeviceDriver A in thread A and B you have to lock that yourself.

closed 2024-08-28 by agmes4