esp-rs/esp-idf #480: ESP32-C3-FH4 How to use gpio12 - 17

cyhuajuan, 2024-09-15:
I know it's not recommended, but I need to use it.

ivmarkov, 2024-09-15:
Then just use it? `esp-idf-*` does not differentiate between the individual c3 variants.

cyhuajuan, 2024-09-15:
> Then just use it? `esp-idf-*` does not differentiate between the individual c3 variants.

Using PinDriver::input(peripherals.pins.gpio14) will cause an infinite reboot.

ivmarkov, 2024-09-15:
> > Then just use it? `esp-idf-*` does not differentiate between the individual c3 variants.
> 
> Using PinDriver::input(peripherals.pins.gpio14) will cause an infinite reboot.

OK. Show me how you would use gpio14 with ESP IDF C calls (or unsafe calls into `esp_idf_sys`) and then we can think how to implement this with `PinDriver`.

In other words, is what you asking for _even possible_, with ESP IDF, let alone with its Rust wrappers?

cyhuajuan, 2024-09-15:
> OK. Show me how you would use gpio14 with ESP IDF C calls (or unsafe calls into `esp_idf_sys`) and then we can think how to implement this with `PinDriver`.
>
> In other words, is what you asking for [...]

I'm also experiencing infinite reboots with esp_idf_sys. As a newcomer, I understand that the datasheet states it's not recommended. I believe there should be a way to use it, so I've raised this issue to inquire.

ivmarkov, 2024-09-15:
You might have better luck asking in the Matrix channel.