esp-rs/esp-idf #482: How to set log level to debug

mindoodoo, 2024-09-25:
Hey guys,

Thanks for the awesome work !

I'm not sure this is the right place to ask, but I cannot for the life of me get my log messages logged using `log::debug!` to display. I tried modifying my sdkconfig file as well as using the `set_target_level()` method, but to no avail.

Any clue ?

Thanks !

elipsitz, 2024-09-26:
What did you add to your `sdkconfig` file? Log level changes are in a specific format (output from menuconfig). See here: https://github.com/esp-rs/esp-idf-template/issues/215#issuecomment-2172912235

Note: be careful increasing log verbosity globally, IIRC esp-idf logs a *lot* at debug level.

ivmarkov, 2024-09-26:
> Note: be careful increasing log verbosity globally, IIRC esp-idf logs a _lot_ at debug level.

Indeed ... and most often than not causes stack overflows.

closed 2024-09-26 by ivmarkov

Florian-Schoenherr, 2026-01-04:
Okay, but how do I not set it globally? I just want my application logs.