esp-rs/esp-idf #491: Incorrect pin assignment for cs in sd/spi.rs on git master branch

jimmer74, 2024-10-11:
Hi, trying to work out why my sdcard wasn't working, I was noodling through the source (i'm using the git [pathc.crates-io] in my Cargo.toml) and found the fololowing for spi_driver::new():

 ```
   pub fn new(
        spi_driver: T,
        cs: Option<impl Peripheral<P = impl OutputPin> + 'd>,
        cd: Option<impl Peripheral<P = impl InputPin> + 'd>,
        wp: Option<impl Peripheral<P = impl InputPin> + 'd>,
        int: Option<impl Peripheral<P = impl InputPin> + 'd>,
        #[cfg(not(any(
            esp_idf_version_major = "4",
            all(esp_idf_version_major = "5", esp_idf_version_minor = "0"),
            all(esp_idf_version_major = "5", esp_idf_version_minor = "1"),
        )))] // For ESP-IDF v5.2 and later
        wp_active_high: Option<bool>,
    ) -> Result<Self, EspError>
    where
        T: Borrow<SpiDriver<'d>>,
    {
        let dev_config = sdspi_device_config_t {
            host_id: spi_driver.borrow().host(),
            gpio_cs: cs.map(|cd| cd.into_ref().deref().pin()).unwrap_or(-1), <----------- this line!!
            gpio_cd: cd.map(|cd| cd.into_ref().deref().pin()).unwrap_or(-1),
            gpio_wp: wp.map(|wp| wp.into_ref().deref().pin()).unwrap_or(-1),
            gpio_int: int.map(|int| int.into_ref().deref().pin()).unwrap_or(-1),
```

Appears the card detect pin is being mapped into the gpio_cs AND gpio_cd. Doesn't seem to effect functionality in my case as only sdcard is on this bus, but might stymie people with multiple spi devices?

Best, Jim

Vollbrecht, 2024-10-11:
oh yeah that looks wrong, do you want to PR a quick fix?

jimmer74, 2024-10-11:
I've actually just realised it's "right" - looking at it again as the gpio.cs is mapped as|cd| in the closure, so it's actually dereffing the right pin anyway, just the captured variable name is wrong!

I've been looking at misbehaving code all day, now it's all starting to look naughty to me!

Vollbrecht, 2024-10-11:
yeah its "right" that is technically correct but well it's still a "typo" overall  ;P

jimmer74, 2024-10-11:
https://github.com/esp-rs/esp-idf-hal/pull/492

1st one I've done, so hoping it's right!

closed 2024-10-11 by Vollbrecht