esp-rs/esp-idf #493: Unable to store ADC handles in a struct

sebastianohl, 2024-10-20:
Hello,

I would like to create a driver that should read an ADC channel. The idea is to create a struct storing the handles in a new function and then use it afterwards. With the legacy ADC driver this was possible due to the fact that there was no direct link between ADCDriver and ADCChannelDriver. However, the new ADCChannelDriver requires a reference to the ADCDriver. If storing this to a struct, the compiler complains that the reference is a local variable.

Do I understand something wrong? Is there a more obvious solution to achieve something similar?

`
pub struct WindDirection<'a> {
    adc_driver: AdcDriver<'a, esp_idf_hal::adc::ADC2>,
    adc_channel:
        AdcChannelDriver<'a, esp_idf_hal::gpio::Gpio12, &AdcDriver<'a, esp_idf_hal::adc::ADC2>>,
}

#[allow(dead_code)]
impl<'a> WindDirection<'a>
/*where
P: esp_idf_hal::gpio::ADCPin,*/
{
    pub fn new(/*pin: P, adc: impl Peripheral<P = P::Adc> + 'a*/) -> Result<Self, EspError> {
        let config = AdcChannelConfig {
            attenuation: DB_11,
            calibration: true,
            ..Default::default()
        };

        let peripherals = Peripherals::take()?;
        let adc_driver = AdcDriver::<'a>::new(peripherals.adc2)?;
        let adc_channel =
            AdcChannelDriver::<'a>::new(&adc_driver, peripherals.pins.gpio12, &config)?;

        Ok(WindDirection { <---- here the compiler complains
            adc_driver,
            adc_channel,
        })
    }

    pub fn measure(&mut self) -> Result<(), EspError> {
        self.adc_driver.read(&mut self.adc_channel)?;
        Ok(())
    }
}
`

The compiler error:
`
error[E0515]: cannot return value referencing local variable `adc_driver`
  --> src/windirection.rs:51:9
   |
49 |               AdcChannelDriver::<'a>::new(&adc_driver, peripherals.pins.gpio12, &config)?;
   |                                           ----------- `adc_driver` is borrowed here
50 |
51 | /         Ok(WindDirection {
52 | |             adc_driver,
53 | |             adc_channel,
54 | |         })
   | |__________^ returns a value referencing data owned by the current function

`

ivmarkov, 2024-10-20:
You are fighting a basic Rust problem, not something related to `esp_idf_hal` in that you are trying to create a self-referential struct, which cannot be expressed in safe Rust today. Rather than using a simple `&` borrow of the driver, y9u can instead wrap it in an `Arc` and then stored the armed driver, as well as pass the armed driver to the channel driver.

sebastianohl, 2024-10-23:
thanks a lot.  It worked even that I don't like to wrap everything in arc or refcells... but it seams that's the way rust handle these things.

closed 2024-10-23 by sebastianohl

ivmarkov, 2024-10-23:
> but it seams that's the way rust handle these things.

Not necessarily. There are other ways, in that you can store in your struct a `&` **reference** to `AdcDriver` together with an owned intance of `AdcChannelDriver` but then you have to introduce a (covariant) lifetime in the struct and deal with that everywhere.