esp-rs/esp-idf #497: Cannot compile in esp32.

UnwhiteCat, 2024-11-22:
Here is the output in stdout:
```bash
UnwhiteCat@uccomputer ~/Codes/rust/hello-world $ cargo build
   Compiling compiler_builtins v0.1.123
   Compiling core v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.158
   Compiling proc-macro2 v1.0.92
   Compiling unicode-ident v1.0.14
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.164
   Compiling serde v1.0.215
   Compiling bitflags v2.6.0
   Compiling regex-syntax v0.8.5
   Compiling crossbeam-utils v0.8.20
   Compiling cfg_aliases v0.2.1
   Compiling glob v0.3.1
   Compiling rustix v0.38.41
   Compiling nix v0.29.0
   Compiling rustversion v1.0.18
   Compiling log v0.4.22
   Compiling linux-raw-sys v0.4.14
   Compiling prettyplease v0.2.25
   Compiling syn v1.0.109
   Compiling minimal-lexical v0.2.1
   Compiling same-file v1.0.6
   Compiling thiserror v1.0.69
   Compiling either v1.13.0
   Compiling serde_json v1.0.133
   Compiling shlex v1.3.0
   Compiling aho-corasick v1.1.3
   Compiling bstr v1.11.0
   Compiling nom v7.1.3
   Compiling walkdir v2.5.0
   Compiling clang-sys v1.8.1
   Compiling cvt v0.1.2
   Compiling libloading v0.8.5
   Compiling crossbeam-epoch v0.9.18
   Compiling ryu v1.0.18
   Compiling quote v1.0.37
   Compiling home v0.5.9
   Compiling itoa v1.0.13
   Compiling crossbeam-deque v0.8.5
   Compiling once_cell v1.20.2
   Compiling bindgen v0.69.5
   Compiling syn v2.0.89
   Compiling heck v0.4.1
   Compiling anyhow v1.0.93
   Compiling itertools v0.12.1
   Compiling cc v1.1.30
   Compiling lazy_static v1.5.0
   Compiling normpath v1.3.0
   Compiling rustc-hash v1.1.0
   Compiling lazycell v1.3.0
   Compiling bitflags v1.3.2
   Compiling fastrand v2.2.0
   Compiling camino v1.1.9
   Compiling semver v1.0.23
   Compiling autocfg v1.4.0
   Compiling ident_case v1.0.1
   Compiling fnv v1.0.7
   Compiling embedded-io-async v0.6.1
   Compiling heapless v0.8.0
   Compiling unicode-xid v0.2.6
   Compiling iana-time-zone v0.1.61
   Compiling embassy-time-driver v0.1.0
   Compiling const_format_proc_macros v0.2.33
   Compiling regex-automata v0.4.9
   Compiling filetime v0.2.25
   Compiling cmake v0.1.51
   Compiling num-traits v0.2.19
   Compiling version_check v0.9.5
   Compiling cexpr v0.6.0
   Compiling which v4.4.2
   Compiling tempfile v3.14.0
   Compiling embedded-hal-async v1.0.0
   Compiling fs_at v0.2.1
   Compiling embassy-sync v0.6.0
   Compiling chrono v0.4.38
   Compiling remove_dir_all v0.8.4
   Compiling uncased v0.9.10
   Compiling litrs v0.4.1
   Compiling document-features v0.2.10
   Compiling strum_macros v0.24.3
   Compiling globset v0.4.15
   Compiling regex v1.11.1
   Compiling ignore v0.4.23
   Compiling darling_core v0.20.10
   Compiling globwalk v0.8.1
   Compiling strum v0.24.1
   Compiling serde_derive v1.0.215
   Compiling thiserror-impl v1.0.69
   Compiling build-time v0.1.3
   Compiling num_enum_derive v0.7.3
   Compiling darling_macro v0.20.10
   Compiling darling v0.20.10
   Compiling enumset_derive v0.10.0
   Compiling rustc-std-workspace-core v1.99.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc)
   Compiling adler v1.0.2
   Compiling rustc-demangle v0.1.24
   Compiling unwind v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/unwind)
   Compiling cargo-platform v0.1.8
   Compiling envy v0.4.2
   Compiling embuild v0.32.0
   Compiling cargo_metadata v0.18.1
   Compiling rustc-std-workspace-alloc v1.99.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-alloc)
   Compiling panic_abort v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_abort)
   Compiling panic_unwind v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_unwind)
   Compiling gimli v0.29.0
   Compiling object v0.36.2
   Compiling hashbrown v0.14.5
   Compiling std_detect v0.1.5 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
   Compiling miniz_oxide v0.7.4
   Compiling addr2line v0.22.0
   Compiling esp-idf-sys v0.35.0
   Compiling esp-idf-hal v0.44.1
   Compiling esp-idf-svc v0.49.1
   Compiling hello-world v0.1.0 (/home/UnwhiteCat/Codes/rust/hello-world)
   Compiling proc_macro v0.0.0 (/home/UnwhiteCat/.rustup/toolchains/esp/lib/rustlib/src/rust/library/proc_macro)
   Compiling byteorder v1.5.0
   Compiling nb v1.1.0
   Compiling embedded-io v0.6.1
   Compiling stable_deref_trait v1.2.0
   Compiling pin-utils v0.1.0
   Compiling pin-project-lite v0.2.15
   Compiling futures-core v0.3.31
   Compiling embedded-hal v1.0.0
   Compiling futures-task v0.3.31
   Compiling critical-section v1.2.0
   Compiling futures-util v0.3.31
   Compiling nb v0.1.3
   Compiling enumset v1.1.5
   Compiling const_format v0.2.33
   Compiling void v1.0.2
   Compiling embedded-hal-nb v1.0.0
   Compiling hash32 v0.3.1
   Compiling embedded-hal v0.2.7
   Compiling embedded-can v0.4.1
   Compiling atomic-waker v1.1.2
   Compiling num_enum v0.7.3
   Compiling embassy-futures v0.1.1
   Compiling embedded-svc v0.28.0
error[E0063]: missing fields `hw_reset_assert_time_us` and `post_hw_reset_delay_ms` in initializer of `esp_idf_hal::sys::eth_phy_config_t`
   --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/eth.rs:806:9
    |
806 |         eth_phy_config_t {
    |         ^^^^^^^^^^^^^^^^ missing `hw_reset_assert_time_us` and `post_hw_reset_delay_ms`

error[E0063]: missing field `rssi_5g_adjustment` in initializer of `esp_idf_hal::sys::wifi_scan_threshold_t`
   --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/wifi.rs:189:24
    |
189 |             threshold: wifi_scan_threshold_t {
    |                        ^^^^^^^^^^^^^^^^^^^^^ missing `rssi_5g_adjustment`

error[E0063]: missing field `duty_cycle_pos` in initializer of `esp_idf_hal::sys::sdspi_device_config_t`
  --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/spi.rs:41:29
   |
41 |         let configuration = sdspi_device_config_t {
   |                             ^^^^^^^^^^^^^^^^^^^^^ missing `duty_cycle_pos`

error[E0063]: missing fields `clk`, `cmd`, `d0` and 7 other fields in initializer of `esp_idf_hal::sys::sdmmc_slot_config_t`
   --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/mmc.rs:148:28
    |
148 |             configuration: sdmmc_slot_config_t {
    |                            ^^^^^^^^^^^^^^^^^^^ missing `clk`, `cmd`, `d0` and 7 other fields

error[E0063]: missing fields `current_limit`, `driver_strength` and `is_slot_set_to_uhs1` in initializer of `esp_idf_hal::sys::sdmmc_host_t`
  --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/host.rs:53:20
   |
53 |         let host = sdmmc_host_t {
   |                    ^^^^^^^^^^^^ missing `current_limit`, `driver_strength` and `is_slot_set_to_uhs1`

error[E0063]: missing fields `current_limit`, `driver_strength` and `is_slot_set_to_uhs1` in initializer of `esp_idf_hal::sys::sdmmc_host_t`
   --> /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/host.rs:125:20
    |
125 |         let host = sdmmc_host_t {
    |                    ^^^^^^^^^^^^ missing `current_limit`, `driver_strength` and `is_slot_set_to_uhs1`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-svc` (lib) due to 6 previous errors
```

ivmarkov, 2024-11-22:
The ESP IDF version you are using and the version of the `esp-idf-*` crates you are using.

UnwhiteCat, 2024-11-22:
The ESP IDF version I used is 5.5.0 and the esp-idf-sys version is 0.35.0 and esp-idf-hal version is 0.44.1 and esp-idf-svc version is 0.49.1.

ivmarkov, 2024-11-22:
There is no such thing as "ESP IDF 5.5.0" yet. Even ESP IDF 5.4 is not released, so you've probably used `master`.
The `esp-idf-template` (if you used it to generated your binary crate) explicitly advises against using ESP IDF `master`. Also in the README.

Please use 5.2 instead. We'll soon support 5.3 too, but not yet.

closed 2024-11-22 by ivmarkov

UnwhiteCat, 2024-11-22:
I have switched the IDF branch from master to release/v5.2. It seams that esp-idf-halt has been perfectly compiled, but there is new wrong came out.
Here is the fail build log in stdout.
```bash
UnwhiteCat@uccomputer ~/Codes/rust/hello-world $ cargo build
   Compiling byteorder v1.5.0
   Compiling nb v1.1.0
   Compiling embedded-io v0.6.1
   Compiling stable_deref_trait v1.2.0
   Compiling futures-task v0.3.31
   Compiling pin-project-lite v0.2.15
   Compiling embedded-hal v1.0.0
   Compiling futures-core v0.3.31
   Compiling pin-utils v0.1.0
   Compiling esp-idf-sys v0.35.0
   Compiling const_format v0.2.33
   Compiling libc v0.2.164
   Compiling enumset v1.1.5
   Compiling void v1.0.2
   Compiling critical-section v1.2.0
   Compiling cfg-if v1.0.0
   Compiling nb v0.1.3
   Compiling num_enum v0.7.3
   Compiling embedded-can v0.4.1
   Compiling serde v1.0.215
   Compiling futures-util v0.3.31
   Compiling atomic-waker v1.1.2
   Compiling log v0.4.22
   Compiling embedded-hal v0.2.7
   Compiling uncased v0.9.10
   Compiling embedded-io-async v0.6.1
   Compiling embedded-hal-async v1.0.0
   Compiling embedded-hal-nb v1.0.0
   Compiling hash32 v0.3.1
   Compiling embassy-time-driver v0.1.0
   Compiling embassy-futures v0.1.1
   Compiling heapless v0.8.0
   Compiling embassy-sync v0.6.0
   Compiling embedded-svc v0.28.0
error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/home/UnwhiteCat/Codes/rust/hello-world/target/debug/build/esp-idf-sys-342aff65c16c3e4c/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:rerun-if-changed=/home/UnwhiteCat/Codes/rust/hello-world/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/UnwhiteCat/Codes/rust/hello-world/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-e6427c24b8fc184c/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/UnwhiteCat/esp/esp-idf" IDF_TARGET="esp32" PATH="/home/UnwhiteCat/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/run/user/1000/fnm_multishells/36450_1732286552774/bin:/home/UnwhiteCat/.cargo/bin:/home/UnwhiteCat/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/run/user/1000/fnm_multishells/36370_1732286544652/bin:/home/UnwhiteCat/.cargo/bin:/home/UnwhiteCat/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/run/user/1000/fnm_multishells/18340_1732286128302/bin:/home/UnwhiteCat/.cargo/bin:/home/UnwhiteCat/esp/esp-idf/components/espcoredump:/home/UnwhiteCat/esp/esp-idf/components/partition_table:/home/UnwhiteCat/esp/esp-idf/components/app_update:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/UnwhiteCat/.espressif/tools/cmake/3.30.2/bin:/home/UnwhiteCat/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/UnwhiteCat/.espressif/tools/ninja/1.12.1/:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/UnwhiteCat/.espressif/tools/cmake/3.30.2/bin:/home/UnwhiteCat/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/UnwhiteCat/.espressif/tools/ninja/1.12.1/:/home/UnwhiteCat/esp/esp-idf/components/espcoredump:/home/UnwhiteCat/esp/esp-idf/components/partition_table:/home/UnwhiteCat/esp/esp-idf/components/app_update:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/home/UnwhiteCat/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/UnwhiteCat/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/UnwhiteCat/.espressif/tools/cmake/3.30.2/bin:/home/UnwhiteCat/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/UnwhiteCat/.espressif/to [... cut ...]
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/UnwhiteCat/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.51/src/lib.rs:1100:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
```
I have coppied the export-esp.sh file's content to my .bashrc file, and the environment should be set properly.