esp-rs/esp-idf #498: "std::io::Read not implemented for UartDriver" anymore?

gb6881, 2024-11-23:
Hi,

unluckily i did an espup update (afaik) and now i'm getting a message i cannot compile my project anymore.
orignally i'm using them sml-rs package, but i've stripped it  down:

`❯ CRATE_CC_NO_DEFAULTS=1  cargo build
   Compiling bindgen v0.69.5
   Compiling embuild v0.32.0
   Compiling esp-idf-sys v0.35.0
   Compiling esp-idf-hal v0.44.1
   Compiling esp-idf-svc v0.49.1
   Compiling test-esp-idf-sys v0.1.0 (/Users/gb/Documents/PRIVATE/esp/test)
error[E0277]: the trait bound `UartDriver<'_>: std::io::Read` is not satisfied
  --> src/main.rs:22:33
   |
22 |         Ok(r ) => { from_reader(r);}
   |                     ----------- ^ the trait `std::io::Read` is not implemented for `UartDriver<'_>`
   |                     |
   |                     required by a bound introduced by this call
   |
note: required by a bound in `from_reader`
  --> src/main.rs:26:23
   |
26 | pub fn from_reader<R: std::io::Read>( _reader: R)  {
   |                       ^^^^^^^^^^^^^ required by this bound in `from_reader`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `test-esp-idf-sys` (bin "test-esp-idf-sys") due to 1 previous error`

test project sources:
[test.zip](https://github.com/user-attachments/files/17880447/test.zip)

build target: xtensa-esp32s3-espidf

output  espup update:
`
[info]: Updating the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/gb/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[warn]: Previous installation of LLVM exists in: '/Users/gb/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/Users/gb/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/gb/.espup/esp-clang'
[warn]: Previous installation of Xtensa Rust 1.82.0.3 exists in: '/Users/gb/.rustup/toolchains/esp'. Reusing this installation
[info]: Update successfully completed!

        To get started, you need to set up some environment variables by running: '. /Users/gb/export-esp.sh'
        This step must be done every time you open a new terminal.
            See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
`
i'm runing a arm mac (mac os 15.1.1), but had the same message also under linux.

any idea what changed, what i can do to fix this problem?

Thanks, Georg

Vollbrecht, 2024-11-23:
Are you sure you are talking about std::io::Read and not some other Read trait? We never implemented a std::io::Read on UartDriver as far as i can see. We provide the embedded_io Read impls alongside with e-hal_nb and e-hal_0.2 impls, and if i check the history of `uart.rs` with `git log -p -- uart.rs | grep "::Read"`` i dont see where there was at any time such a implementation?

ivmarkov, 2024-11-23:
I also think we have never implemented std::io::Read. There are adapters that adapt embedded_io::Read to std::io::Read and that's why I think we never implemented those.

gb6881, 2024-11-24:
Thanks for you quick reply!

then i'll have to look, from which dependency this implementation vanished :)

closed 2024-11-24 by gb6881