esp-rs/esp-idf #500: ESP32-H2 missing GPIO pins above 20

bitmagier, 2024-12-13:
The lib exposes GPIO Pins 0..20 for the ESP32-H2 GPIO.
As one can see in section "2.1 Pin Layout" in the [Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-h2_datasheet_en.pdf), there are also the following hardware pins available to use:
Pins: 22, 25, 26, 27

Especially pin 22 and pin 25 are extremely useful, because they are also exposed in the ESP32-H2-WROOM-03 model (which is also used by me).

I suggest to simply add these GPIO pin numbers in esp-idf-hal.

ivmarkov, 2024-12-13:
I don't have a h2 around, so what would be most efficient - if you don't mind - is if you could actually open a PR adding those. This way you could also test if these pins do actually work.

bitmagier, 2024-12-13:
.. I'm just preparing one.

bitmagier, 2024-12-13:
https://github.com/esp-rs/esp-idf-hal/pull/501

bitmagier, 2024-12-13:
Unfortunately the `main` build seems to be broken at the moment. I get the same compile-errors locally as the pipeline shows on github.

ivmarkov, 2024-12-13:
> Unfortunately the `main` build seems to be broken at the moment. I get the same compile-errors locally as the pipeline shows on github.

Use a slightly older `nightly` or the ESP toolchain. What is broken is upstream STD in `nightly` since ~ 2 days.

bitmagier, 2024-12-13:
Thanks, my local build works.
Validating pin functionality on hardware-level is the next best thing to to.
Will need some time.

bitmagier, 2024-12-16:
A added also pins 23+24 (used by Serial debug interface)

I successfully verified hardware functionality for pins 22 and 25.
The other 4 added pins should work too. Since they are not directly recommended to be used, one may play around with them by himself.

bitmagier, 2024-12-16:
PR https://github.com/esp-rs/esp-idf-hal/pull/501 is updated, works for me and is ready to be pulled.

closed 2024-12-18 by bitmagier