esp-rs/esp-idf #513: Naming confliction when using esp32-camera as extra component

ethe, 2025-02-02:
## Bug description

1. `git clone https://github.com/espressif/esp32-camera.git` to components folder
2. specify extra component in `Cargo.toml`:
```toml
[[package.metadata.esp-idf-sys.extra_components]]
component_dirs = ["components"]
bindings_header = "src/bindings.h"
```
3. build
```
❯ cargo build --release
   Compiling esp-idf-sys v0.36.1
   Compiling esp-idf-hal v0.45.2
   Compiling esp-idf-svc v0.51.0
   Compiling esp-rs v0.1.0 (/Users/gwo/Idea/esp-rs)
error[E0530]: function parameters cannot shadow statics
  --> /Users/gwo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.45.2/src/adc.rs:70:13
   |
3  | use esp_idf_sys::*;
   |     -------------- the static `resolution` is imported here
...
70 |     fn from(resolution: Resolution) -> Self {
   |             ^^^^^^^^^^ cannot be named the same as a static

error[E0530]: function parameters cannot shadow statics
  --> /Users/gwo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.45.2/src/can.rs:93:17
   |
65 |     use esp_idf_sys::*;
   |         -------------- the static `resolution` is imported here
...
93 |         fn from(resolution: Timing) -> Self {
   |                 ^^^^^^^^^^ cannot be named the same as a static

For more information about this error, try `rustc --explain E0530`.
error: could not compile `esp-idf-hal` (lib) due to 2 previous errors

```

## Expected behavior

Build pass

## Environment

```toml
[[package]]
name = "esp-idf-hal"
version = "0.45.2"
```

Vollbrecht, 2025-02-02:
We by default do not modularise the imports, but put everything into one big bindgen.rs file. If you know import something that is defined with the same name in two different headers that are distinct you get an error like this, For extra components i would suggest to import it as a seperate module. E.g make it available under esp_idf_sys::camera so it does not conflict. The description how you can do it is found [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#extra-esp-idf-components) E.g the bindings_module definition.

ethe, 2025-02-03:
`bindings_module` solved this, thanks

closed 2025-02-03 by ethe