esp-rs/esp-idf #518: Feature request: implement fade features for ledc

justbuchanan, 2025-02-24:
Hardware support for fading leds is neat and I'd like to use it in one of my projects.

If nobody is currently working on this, I can take a look at implementing it.

The fade feature in esp32 hardware is described here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/ledc.html#change-pwm-duty-cycle-using-hardware.

justbuchanan, 2025-02-24:
cc @sirhcel since you implemented the ledc driver initially in https://github.com/esp-rs/esp-idf-hal/pull/43. Please let me know if you have any thoughts/suggestions/advice.

sirhcel, 2025-02-25:
Great to see you picking this up @justbuchanan! I have to admit that I have not looked into fading yet. You could keep me in the loop when opening a PR and I will have a look at it.

ivmarkov, 2025-03-13:
Shall I close?

justbuchanan, 2025-03-13:
Yeah this can be closed now. Thanks!

closed 2025-03-13 by justbuchanan