esp-rs/esp-idf #522: Task Cleanup leaks 160B of Memory

smallbraingames, 2025-03-14:
## Bug description

If a FreeRTOS task is created with ThreadSpawnConfig, it leaks 160 bytes in memory when cle

- Would you like to work on a fix? [y/n]

## To Reproduce

1. Create any task with threadspawnconfig, then join and drop it
2. Check the memory usage before and after

```
 loop {
        log::info!("\n");
        log::info!("============== BEFORE TEST TASK ===========");
        print_memory_info();
        let test_task = spawn("test\0", 1024 * 10, 1, Some(Core::Core0), move || {
            sleep(Duration::from_secs(3));
        });
        test_task.join().unwrap();
        log::info!("\n");
        log::info!("============== AFTER TEST TASK ===========");
        print_memory_info();
    }
```

and in print_memory_info,

```
 log::info!(
            " Free: {} bytes",
            heap_caps_get_free_size(MALLOC_CAP_INTERNAL)
        );
```

## Expected behavior

It to clean up the task fully

## Environment

- esp-idf-svc 0.51
- Target device (MCU): esp32s3

ivmarkov, 2025-03-15:
Are you sure that you are observing a memory leak caused by spawning (`std::thread::spawn`) the new thread, and NOT observing the allocation of the thread-local memory for the `ThreadSpawnConfig` keys itself?

Two experiments you could do to confirm/deny:
- Experiment 1: Do _not_ call `spawn` but still call the `ThreadSpawnConfig` setup. If you still see a memory "leak", you are observing the `ThreadSpawnConfig` local storage allocation in heap, rather than anything else.
- Experiment 2: Do what you do now, but offload the `ThreadSpawnConfig` as well the subsequent `std::thread::spawn` call **in another, "outer" thread**, which is also spawned with `std::thread::spawn`, but does not have a `ThreadSpawnConfig` setup. Once both threads are joined, you should not see a "memory leak" anymore, as the `ThreadSpawnConfig` local storage of the outer thread should die when this thread is joined.

benhansen-io, 2025-05-30:
I think I am seeing the same or similar issue with plain std::thread::spawn

Reposting from the closed esp-rs/rust issue:

```rust
    for i in 0..1000 {
        let handle = std::thread::Builder::new()
            .stack_size(6000)
            .spawn(move || {
                // No op
            })
            .unwrap();
        handle.join().unwrap();
        memory::print_mem_info(&format!("after {i}"));
    }
```

I expected to be able to spawn and then join a thread to reclaim all memory.

The loop is unable to finish. Each iteration consumes about 150 bytes that is not freed. Eventually the device runs out of memory.

Logs:
```
2025-05-30 15:40:14 I (11073) ten_os::commands: executing command: test
2025-05-30 15:40:14 I (11073) rust_boppo::memory: HEAP: 450.1/2419k (19%) DMA: 258.6/317k (81%) Stack HWM: 25848: after 0
2025-05-30 15:40:14 I (11083) rust_boppo::memory: HEAP: 450.2/2419k (19%) DMA: 258.7/317k (81%) Stack HWM: 25848: after 1
2025-05-30 15:40:14 I (11083) rust_boppo::memory: HEAP: 450.4/2419k (19%) DMA: 258.9/317k (82%) Stack HWM: 25848: after 2
2025-05-30 15:40:14 I (11083) rust_boppo::memory: HEAP: 450.6/2419k (19%) DMA: 259.0/317k (82%) Stack HWM: 25848: after 3
2025-05-30 15:40:14 I (11093) rust_boppo::memory: HEAP: 450.7/2419k (19%) DMA: 259.2/317k (82%) Stack HWM: 25848: after 4
....
....
2025-05-30 15:40:15 I (12333) rust_boppo::memory: HEAP: 504.7/2415k (21%) DMA: 313.1/313k (100%) Stack HWM: 25816: after 350
2025-05-30 15:40:15 I (12333) rust_boppo::memory: HEAP: 504.8/2415k (21%) DMA: 313.1/313k (100%) Stack HWM: 25816: after 351
2025-05-30 15:40:15 I (12343) rust_boppo::memory: HEAP: 505.0/2415k (21%) DMA: 313.1/313k (100%) Stack HWM: 25816: after 352
2025-05-30 15:40:15 I (12343) rust_boppo::memory: HEAP: 505.1/2415k (21%) DMA: 313.1/313k (100%) Stack HWM: 25816: after 353
2025-05-30 15:40:15 I (12343) rust_boppo::memory: HEAP: 505.3/2415k (21%) DMA: 313.1/313k (100%) Stack HWM: 25816: after 354
2025-05-30 15:40:15 E (12353) pthread: Failed to create task!
2025-05-30 15:40:15
2025-05-30 15:40:15 thread '<unnamed>' panicked at src/commands.rs:134:14:
2025-05-30 15:40:15 called `Result::unwrap()` on an `Err` value: Os { code: 12, kind: OutOfMemory, message: "Not enough space" }
2025-05-30 15:40:15 note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
2025-05-30 15:40:15
2025-05-30 15:40:15 abort() was called at PC 0x420aa5de on core 0
--- 2025-05-30 15:40:15 0x420aa5de: panic_abort::__rust_start_panic::abort at ??:?
````

### Meta

`rustc --version --verbose`:
```
rustc 1.86.0-nightly (5944fc5cd 2025-04-01) (1.86.0.0)
binary: rustc
commit-hash: 5944fc5cd231b5aaf09b471def70ecefefa4737d
commit-date: 2025-04-01
host: x86_64-unknown-linux-gnu
release: 1.86.0-nightly
LLVM version: 19.1.2
```

Vollbrecht, 2025-06-04:
you run out of memory in under 1 sec with this approach right? 

I think before trying to investigate to deep, i would test to a) allocate a bunch of threads b) drop them c) go to sleep ( such that you give the 0 priority idle thread a change to run) d) check again.

The idle thread does a lot of cleanup, and in your example it has never a change to run because the system crashes after 1 sec.

benhansen-io, 2025-06-04:
Gave that a shot, no real change. Still seeing unbounded memory growth.

Code:

```
    for batch_num in 0..1000 {
        let num_threads_per_batch = 100;
        for i in 0..num_threads_per_batch {
            let handle = std::thread::Builder::new()
                .stack_size(6000)
                .spawn(move || {
                    // do nothing
                })
                .unwrap();
            handle.join().unwrap();
            let thread_num = batch_num * num_threads_per_batch + i;
            memory::print_mem_info(&format!("after {thread_num}"));
        }
        std::thread::sleep(std::time::Duration::from_millis(5000));
    }
```

Output
```
2025-06-04 11:24:17 I (11433) ten_os::commands: executing command: test
2025-06-04 11:24:17 I (11443) rust_boppo::memory: HEAP: 448.9/2419k (19%) DMA: 258.4/317k (81%) Stack HWM: 25960: after 0
2025-06-04 11:24:17 I (11443) rust_boppo::memory: HEAP: 449.1/2419k (19%) DMA: 258.6/317k (81%) Stack HWM: 25960: after 1
2025-06-04 11:24:17 I (11443) rust_boppo::memory: HEAP: 449.2/2419k (19%) DMA: 258.7/317k (81%) Stack HWM: 25960: after 2
2025-06-04 11:24:17 I (11453) rust_boppo::memory: HEAP: 449.4/2419k (19%) DMA: 258.9/317k (81%) Stack HWM: 25960: after 3
....
2025-06-04 11:24:17 I (11773) rust_boppo::memory: HEAP: 464.5/2418k (19%) DMA: 273.9/316k (87%) Stack HWM: 25960: after 99
2025-06-04 11:24:22 I (16773) rust_boppo::memory: HEAP: 464.5/2418k (19%) DMA: 274.0/316k (87%) Stack HWM: 25960: after 100
...
2025-06-04 11:24:33 I (27683) rust_boppo::memory: HEAP: 504.3/2415k (21%) DMA: 313.2/313k (100%) Stack HWM: 25960: after 355
2025-06-04 11:24:33 I (27693) rust_boppo::memory: HEAP: 504.5/2415k (21%) DMA: 313.2/313k (100%) Stack HWM: 25960: after 356
2025-06-04 11:24:33 E (27693) pthread: Failed to create task!
2025-06-04 11:24:33
2025-06-04 11:24:33 thread '<unnamed>' panicked at src/commands.rs:136:18:
2025-06-04 11:24:33 called `Result::unwrap()` on an `Err` value: Os { code: 12, kind: OutOfMemory, message: "Not enough space" }
2025-06-04 11:24:33 note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
2025-06-04 11:24:33
2025-06-04 11:24:33 abort() was called at PC 0x420bb712 on core 0
```

benhansen-io, 2025-06-04:
Let me know if I can help try anything else.

Maybe raw pthread create to see if it leaks? https://docs.esp-rs.org/esp-idf-svc/esp_idf_sys/fn.pthread_create.html

Vollbrecht, 2025-06-04:
yeah technically that is what the std library is calling through newlib, but of course you can try to have a go.

Vollbrecht, 2025-06-04:
In terms of "where all the code lives" You can have a look at the newlib part of the rust libc [here](https://github.com/rust-lang/libc/tree/main/src/unix/newlib). That enables the biggest part of the rust std support. 

Inside the the std library itself its not much if you look [here](https://github.com/rust-lang/rust/tree/master/library/std/src/os/espidf)

Vollbrecht, 2025-06-04:
to close the cycle on the C side, and to answer the question what newlib version is used. It depends on esp-idf / ROM version.

All rom versions can be found [here](https://github.com/espressif/newlib-esp32), and for the rest you can look inside the esp-idf newlib dir in esp-idf version you are using for example [here](https://github.com/espressif/esp-idf/tree/master/components/newlib/src) for esp-idf master

ivmarkov, 2025-06-04:
Then it must be something that the STD wrapper I referenced above does, which your raw `pthread` calls don't.

When debugging such problems, I've found the easiest to be to just comment/modify bits and pieces of code in the STD itself (you have its sources in the active toolchain) and then do `cargo clean` prior to `cargo rebuild` so that the changes to STD are picked up.

(To make sure you are modifying the correct STD sources, deliberately put an erroneous code in there, in the STD `thread.rs` imnpl you would be modifying, so that you could see the compiler bailing out.)

benhansen-io, 2025-06-04:
Ok. Thanks for the pointers on how to test std code.

When I comment out [set_current](https://github.com/rust-lang/rust/blob/d9a739363878a06633eb912e44f02a3c5d6ba08c/library/std/src/thread/mod.rs#L544) inside thread spawn the leak goes away.

The [code](https://github.com/rust-lang/rust/blob/d9a739363878a06633eb912e44f02a3c5d6ba08c/library/std/src/thread/current.rs#L126) for set_current uses thread locals and says:

```
    // Make sure that `crate::rt::thread_cleanup` will be run, which will
    // call `drop_current`.
    crate::sys::thread_local::guard::enable();
    CURRENT.set(thread.into_raw().cast_mut());
```

Does esp-idf clean up thread_locals differently?

benhansen-io, 2025-06-19:
Reporting back that I haven't found time and don't think I am going to be able to do so in the next few months. That said if anyone in the community is interested in working on this in a sponsored capacity, feel free to send me an email with a proposal.

> Side topic: since you mentioned SPIRAM and entering production... I really do hope you have not registered SPIRAM with malloc in ESP-IDF. https://github.com/esp-rs/esp-hal/issues/2027.

Thanks for the FYI. We actually discussed this on esp-rs matrix.org room a few months ago. My project is setup such that only large allocations are going into SPIRAM and I haven't seen any issues from it in testing over the last year+. It will be nice when its fixed upstream though so I appreciate your efforts there.

ivmarkov, 2025-07-22:
> On the other hand, the Rust `std` explicitly does not _guarantee_ that thread cleanup happens, and so from a correctness standpoint this memory leak is okay but suboptimal.

Can you point me to where they state that?
My concern is that we are leaking 160B even for the simplest of threads, where we don't have any _explicit_ (as in - app-defined) thread locals.

Even if Rust does not guarantee thread cleanup, the above behavior might be very, very surprising for users when they spawn threads without any "visible" thread locals.

ivmarkov, 2025-07-22:
But I agree - based on your explanations - that @joboet's impl is likely platform specific and thus incorrect (that is, for ESP-IDF _at least_).

ivmarkov, 2025-07-22:
> > Yes but what you quote is different: they basically say "there are caveats with running TLS destructors when the program is about to exit".
> 
> That is what you're _inferring_ based on the list of known caveats but they never state that this is an exhaustive list, just that they're the currently known ones, and they explicitly state that running destructors is simply "best effort." And even if they _were_ to treat it as an exhaustive list, that just means that they could add this caveat now that it's known.
> 

Yes, I **am** inferring because the STD documentation is way too vague.

> I still think that it's something that should clearly be fixed, of course, 

100% :)

> but it seems entirely acceptable based on the documented requirements.

Exactly. But then bringing your argument to the extreme, a **no-op** thread-local destructor impl would also be completely "compliant", isn't it? 
I doubt this is what they really meant hence why I say this documentation is vague. :)

ivmarkov, 2025-07-22:
I would still be super-interested to hear what @joeboet would say, but @kyrias what do you think w.r.t. fixing it?
One option - I guess - is - we can always upstream a `#[cfg(target_os = "espidf")]` chunk of code if upstream STD - for whatever reason - would like to keep the current impl for the "main" OSes (libcs, rather).

joboet, 2025-07-22:
This is a soundness bug in ESP-IDFs implementation of pthread keys.

According to POSIX,
> Both pthread_getspecific() and pthread_setspecific() may be called from a thread-specific data destructor function. A call to pthread_getspecific() for the thread-specific data key being destroyed shall return the value NULL, unless the value is changed (after the destructor starts) by a call to pthread_setspecific(). Calling pthread_setspecific() from a thread-specific data destructor routine may result either in lost storage (after at least PTHREAD_DESTRUCTOR_ITERATIONS attempts at destruction) or in an infinite loop.[^1]

with `PTHREAD_DESTRUCTOR_ITERATIONS` having a value of at least four[^2].

Thus, `std` is justified in assuming that `pthread_setspecific` is sound to call in a destructor and that calling it with a non-null value will lead to the destructor being called again in the next round of key destruction (at least for three or more destructor rounds). This is necessary to delay the destruction of the `Thread` handle obtainable through `thread::current` until after all other Rust TLS destructors have run.

But both of these assumptions are not fulfilled by ESP-IDF: neither are the four rounds of destruction [advertised by `PTHREAD_DESTRUCTOR_ITERATIONS`](https://github.com/espressif/newlib-esp32/blob/007ee8a2eb5b60a69dfbdb4bf2399413b46a0b6b/newlib/libc/sys/rtems/include/limits.h#L107) actually performed, nor is `pthread_getspecific`/`pthread_setspecific` safe to be called from a destructor:

https://github.com/espressif/ESP8266_RTOS_SDK/blob/d412ac601befc4dd024d2d2adcfaa319c7463e36/components/pthread/pthread_local_storage.c#L144-L152

The list of pthread keys is traversed (only once!) and freed before its TLS slot is reset, leading to use-after-free if `pthread_getspecific`/`pthread_setspecific` is called in a destructor, as that will attempt to traverse the already freed list.

There is unfortunately no correct way for `std` to work around this bug. `std` fundamentally assumes that `pthread_setspecific` is callable in destructors since it makes TLS variables safe to access. Thus any platform-specific workaround for the leaking issue would be only a stopgap solution.

[^1]: https://pubs.opengroup.org/onlinepubs/9799919799/functions/pthread_getspecific.html
[^2]: https://pubs.opengroup.org/onlinepubs/9799919799/basedefs/limits.h.html

joboet, 2025-07-22:
> Okay, we believe that we've found the root cause which appears to be the Rust `std` depending on semantics that POSIX does not guarantee. It seems that `std` is ([here](https://github.com/esp-rs/rust/blob/5944fc5cd231b5aaf09b471def70ecefefa4737d/library/std/src/sys/thread_local/guard/key.rs#L33-L60)) implicitly assuming that _all_ TLS key destructors will be called before _any_ TLS keys without destructors are freed, by using a kind of sentinel thread local with a destructor that then tries to free up the state of TLS keys without destructors.
> 
> POSIX does not appear to guarantee this behavior[¹](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_key_create.html), it only guarantees that all non-NULL keys will have their destructors called at least once, for the purposes of freeing the backing storage of the TLS key that the destructor belongs to. The order of destructors is explicitly unspecified, and the order that TLS keys without destructors are freed in is implicitly unspecified.
> 
> Meanwhile, the ESP-IDF pthreads implementation frees all TLS keys in order[²](https://github.com/espressif/esp-idf/blob/bfe5caf58f742fd35c023335f475114a5b88761e/components/pthread/pthread_local_storage.c#L130-L147), whether or not the key had a destructor.
> 
> As far as I can tell the ESP-IDF implementation is correct, and if this works on any other pthreads implementation it's just by accident. I haven't looked into whether the non-key-based TLS implementation is correct. On the other hand, the Rust `std` explicitly does not _guarantee_ that thread cleanup happens, and so from a correctness standpoint this memory leak is okay but suboptimal.
> 
> This regression appears to have been introduced in [rust-lang/rust#127912](https://github.com/rust-lang/rust/pull/127912) ([commit](https://github.com/esp-rs/rust/commit/d868fdce6b9ddef6abcc8de86b3ba8459def36a2)).
> 
> CC: [@joboet](https://github.com/joboet)

That is definitely the cause of this problem. Though I disagree about ESP-IDFs behaviour being POSIX-compliant. POSIX specifies that
> At thread exit, if a key value has a non-NULL destructor pointer, and the thread has a non-NULL value associated with that key, the value of the key is set to NULL, and then the function pointed to is called with the previously associated value as its sole argument.

I interpret this as meaning that the value of a key is *only* set to NULL *if* the key has a destructor – but the specification is obviously not very clear on this issue. Regardless, ESP-IDF isn't the only platform with this behaviour, macOS does the same, if I remember correctly, but we don't encounter the issue there since it uses a different destructor mechanism.

[^1]: https://pubs.opengroup.org/onlinepubs/9799919799/functions/pthread_key_create.html

kyrias, 2025-07-22:
> I interpret this as meaning that the value of a key is only set to NULL if the key has a destructor – but the specification is obviously not very clear on this issue. Regardless, ESP-IDF isn't the [...]

That section is only specifying the behavior explicitly in the presence of a destructor being specified, it can't be taken to imply anything about the behavior when there is no destructor.  While I agree that it's definitely underspecified, as far as I can tell any ESP-IDF's implementation is correct.

---

Now, maybe Espressif would be willing to switch to a two-stage cleanup routine where it would first loop over the list of TLS keys until there are none remaining with a destructor specified and only once that's done free all remaining ones?  It would be slightly more complex, and would entail more list iterations, but they might find it an acceptable solution if we have a good reason for requesting it.

If so it would solve it for the Espressif Xtensa chips at least, though it feels kind of unsatisfying that it could still be an issue for other targets, even if technically speaking this memory leak would be allowed by the documentation.  Solving it entirely in `std` for these kinds of targets seems somewhat complicated though since I think it would probably require making the `CURRENT` thread local hold different types depending on the underlying implementation, with the two-stage destructor on the actual thread-local that it tries to free the resources of?

Or maybe a solution using a tagged pointer in the thread local could work, leaving the underlying type the same?  Not sure.

On the other hand, it might not be entirely unreasonable to otherwise just document this as a requirement `std` has of the platforms it supports, at least if Espressif would be amenable to the change.

ivmarkov, 2025-07-22:
@kyrias Even if the Espressif folks are open to change to a two-stage TLS cleanup, that might take months, and it is unclear whether they would back-port it to older (but still supported) ESP-IDF releases.

Implementing an alternative, ESP-IDF-specific code in Rust STD has a much higher chance of quick upstreaming. And we absolutely (probably even urgently) need that, given that currently, **every** single thread leaks. So I would say, we shpould pursue that path instead, at least for the short-term.

JacksonUtsch, 2025-07-23:
I seem to be having the same issue. Appreciate the progress for the issue.