esp-rs/esp-idf #523: Reconfigure WiFi with static IP after initialization

br0kenpixel, 2025-03-30:
My project stores a list of APs and tries to connect to the best one. I'd like to implement a feature that allows setting a static IP for **some** of those networks, while relying on DHCP for others.

When I first initialize `EspWifi` I'm not doing any static IP setup. However, I have a separate `connect()` method for connecting to a specific network. I'd like to "reconfigure" the STA interface with a static IP configuration.

This is essentially my WiFi initialization code:
```rust
let wifi = WifiDriver::new(modem, sys_loop, None)?;
esp!(unsafe { esp_wifi_set_storage(wifi_storage_t_WIFI_STORAGE_RAM) })?;

let mut wifi = EspWifi::wrap_all(wifi, EspNetif::new_with_conf(&ip_config)?)?;
wifi.set_configuration(&Configuration::Client(ClientConfiguration::default()))?;

wifi.start()?;
esp!(unsafe { esp_wifi_set_country_code(WIFI_COUNTRY_CODE.as_ptr().cast(), true) })?;
```

My `connect()` method is the following:
```rust
pub fn connect(&mut self, ssid: &str, psk: &str, auth: AuthMethod, static_ip_config: Option<NetifConfiguration>, timeout: Duration) -> OsResult<()> {
    if let Some(config) = static_ip_config {
        // ESP_ERR_INVALID_ARG
        self.driver.swap_netif_sta(EspNetif::new_with_conf(&config)?)?;
    }

    // ...
}
```

However, `swap_netif_sta()` returns `ESP_ERR_INVALID_ARG`. I tried calling `EspWifi::stop()` before `swap_netif_sta()`, but I get the same error. Is something like this even possible?

If I initialize `EspWifi` with static configuration, it works, but I can't do that because at that point I don't know which network I'll actually connect to. My app needs to perform a scan first and then do some filtering. This would require removing support for multiple APs and only relying on a single one.

Vollbrecht, 2025-03-30:
I am not sure out of my head why that would not work, though one alternative think you always can try is, instead of wifi_driver the modem as a owned value you can pass it by &mut modem. Next step is when you now want to change the driver config, you can drop the wifi, and  then you are free to recreate again via &mut modem. That should at least always work.

br0kenpixel, 2025-04-05:
Yes that would also be an option, but this should work too. All I could figure out is that `swap_netif_sta` is not actually the problem, but `EspNetif::new_with_conf` is failing. More specifically, the call to `esp_netif_new` [here](https://github.com/esp-rs/esp-idf-svc/blob/65fdf52c09aa8516d3198813d2a311071a344624/src/netif.rs#L431) is returning `NULL`.

ivmarkov, 2025-04-05:
> Yes that would also be an option, but this should work too. All I could figure out is that `swap_netif_sta` is not actually the problem, but `EspNetif::new_with_conf` is failing. More specifically, [...]

`esp_netif_new` is open source so you could look in the ESP-IDF C source of that function to figure out why it is returning null.