esp-rs/esp-idf #526: cannot find struct, variant or union type `twai_timing_config_t__bindgen_ty_1` in this scope

syx-413, 2025-05-13:
Something went wrong ?
my .cargo/config.toml
```toml
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
rustflags = [ "--cfg",  "espidf_time64"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "master"
ESP_IDF_TOOLS_INSTALL_DIR = { value = "global" }
```
build error
```
s413@Shaox ~/e/esp-idf (master)> . /home/s413/esp/esp-idf/export.fish
Checking "python3" ...
Python 3.13.3
python3 has been detected
Activating ESP-IDF 5.5
Setting IDF_PATH to '/home/s413/esp/esp-idf'.
* Checking python version ... 3.13.3
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... fish
* Detecting outdated tools in system ... OK - no outdated tools found
* Shell completion ... OK

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
s413@Shaox ~/e/esp-idf (master)> idf.py   --version
ESP-IDF v5.5-dev-3598-g7cf5dacd4a-dirty
s413@Shaox ~/e/esp-idf (master)> cdh
 d  4)  ~
 c  3)  ~/Embbed/esp32/esp53
 b  2)  ~/Embbed/esp32
 a  1)  ~/Embbed/esp32/esp32blink
Select directory by letter or number: a
s413@Shaox ~/E/e/esp32blink (master)> cargo clean && rm -rf .embuild && cargo build
     Removed 3583 files, 1.8GiB total
   Compiling compiler_builtins v0.1.140
   Compiling core v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.169
   Compiling proc-macro2 v1.0.95
   Compiling unicode-ident v1.0.18
   Compiling object v0.36.7
   Compiling std v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.172
   Compiling serde v1.0.219
   Compiling bitflags v2.9.0
   Compiling regex-syntax v0.8.5
   Compiling crossbeam-utils v0.8.21
   Compiling glob v0.3.2
   Compiling cfg_aliases v0.2.1
   Compiling nix v0.29.0
   Compiling rustversion v1.0.20
   Compiling syn v1.0.109
   Compiling log v0.4.27
   Compiling aho-corasick v1.1.3
   Compiling prettyplease v0.2.32
   Compiling clang-sys v1.8.1
   Compiling bstr v1.12.0
   Compiling either v1.15.0
   Compiling crossbeam-epoch v0.9.18
   Compiling quote v1.0.40
   Compiling minimal-lexical v0.2.1
   Compiling shlex v1.3.0
   Compiling thiserror v1.0.69
   Compiling syn v2.0.101
   Compiling rustix v0.38.44
   Compiling getrandom v0.3.3
   Compiling rustix v1.0.7
   Compiling same-file v1.0.6
   Compiling serde_json v1.0.140
   Compiling walkdir v2.5.0
   Compiling nom v7.1.3
   Compiling crossbeam-deque v0.8.6
   Compiling libloading v0.8.7
   Compiling cvt v0.1.2
   Compiling once_cell v1.21.3
   Compiling linux-raw-sys v0.4.15
   Compiling linux-raw-sys v0.9.4
   Compiling heck v0.4.1
   Compiling ryu v1.0.20
   Compiling itoa v1.0.15
   Compiling anyhow v1.0.98
   Compiling bindgen v0.71.1
   Compiling regex-automata v0.4.9
   Compiling cc v1.2.22
   Compiling cexpr v0.6.0
   Compiling fs_at v0.2.1
   Compiling itertools v0.13.0
   Compiling home v0.5.11
   Compiling normpath v1.3.0
   Compiling fastrand v2.3.0
   Compiling rustc-hash v2.1.1
   Compiling bitflags v1.3.2
   Compiling which v4.4.2
   Compiling tempfile v3.20.0
   Compiling remove_dir_all v0.8.4
   Compiling cmake v0.1.54
   Compiling globset v0.4.16
   Compiling regex v1.11.1
   Compiling filetime v0.2.25
   Compiling ident_case v1.0.1
   Compiling strsim v0.11.1
   Compiling semver v1.0.26
   Compiling camino v1.1.9
   Compiling fnv v1.0.7
   Compiling ignore v0.4.23
   Compiling autocfg v1.4.0
   Compiling heapless v0.8.0
   Compiling embedded-io-async v0.6.1
   Compiling litrs v0.4.1
   Compiling darling_core v0.20.11
   Compiling strum_macros v0.24.3
   Compiling globwalk v0.8.1
   Compiling num-traits v0.2.19
   Compiling embassy-time-queue-utils v0.1.0
   Compiling embassy-time-driver v0.2.0
   Compiling unicode-xid v0.2.6
   Compiling iana-time-zone v0.1.63
   Compiling const_format_proc_macros v0.2.34
   Compiling document-features v0.2.11
   Compiling version_check v0.9.5
   Compiling embedded-hal-async v1.0.0
   Compiling chrono v0.4.41
   Compiling embassy-sync v0.6.2
   Compiling embassy-executor v0.7.0
   Compiling uncased v0.9.10
   Compiling strum v0.24.1
   Compiling serde_derive v1.0.219
   Compiling thiserror-impl v1.0.69
   Compiling num_enum_derive v0.7.3
   Compiling darling_macro v0.20.11
   Compiling build-time v0.1.3
   Compiling darling v0.20.11
   Compiling enumset_derive v0.11.0
   Compiling embassy-executor-macros v0.6.2
   Compiling rustc-std-workspace-core v1.99.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc)
   Compiling unwind v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/unwind)
   Compiling adler v1.0.2
   Compiling rustc-demangle v0.1.24
   Compiling cargo-platform v0.1.9
   Compiling envy v0.4.2
   Compiling rustc-std-workspace-alloc v1.99.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-alloc)
   Compiling panic_abort v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_abort)
   Compiling panic_unwind v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_unwind)
   Compiling gimli v0.29.0
   Compiling hashbrown v0.15.2
   Compiling std_detect v0.1.5 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
   Compiling miniz_oxide v0.7.4
   Compiling embuild v0.33.0
   Compiling cargo_metadata v0.18.1
   Compiling addr2line v0.22.0
   Compiling esp-idf-sys v0.36.1
   Compiling esp-idf-hal v0.45.2
   Compiling esp-idf-svc v0.51.0
   Compiling esp32blink v0.1.0 (/home/s413/Embbed/esp32/esp32blink)
   Compiling proc_macro v0.0.0 (/home/s413/.rustup/toolchains/esp/lib/rustlib/src/rust/library/proc_macro)
   Compiling byteorder v1.5.0
   Compiling stable_deref_trait v1.2.0
   Compiling nb v1.1.0
   Compiling critical-section v1.2.0
   Compiling embedded-io v0.6.1
   Compiling embedded-hal v1.0.0
   Compiling futures-task v0.3.31
   Compiling pin-project-lite v0.2.16
   Compiling pin-utils v0.1.0
   Compiling futures-core v0.3.31
   Compiling hash32 v0.3.1
   Compiling nb v0.1.3
   Compiling futures-util v0.3.31
   Compiling void v1.0.2
   Compiling futures-sink v0.3.31
   Compiling const_format v0.2.34
   Compiling enumset v1.1.6
   Compiling embedded-hal v0.2.7
   Compiling embedded-hal-nb v1.0.0
   Compiling embedded-can v0.4.1
   Compiling num_enum v0.7.3
   Compiling atomic-waker v1.1.2
   Compiling futures-io v0.3.31
   Compiling embassy-futures v0.1.1
   Compiling embedded-svc v0.28.1
error[E0422]: cannot find struct, variant or union type `twai_timing_config_t__bindgen_ty_1` in this scope
     --> /home/s413/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-hal-0.45.2/src/can.rs:179:39
      |
179   |                     __bindgen_anon_1: twai_timing_config_t__bindgen_ty_1 { triple_sampling },
      |                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a struct with a similar name exists: `twai_general_config_t__bindgen_ty_1`
      |
     ::: /home/s413/Embbed/esp32/esp32blink/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-5fd569abb3384bc0/out/bindings.rs:89861:1
      |
89861 | pub struct twai_general_config_t__bindgen_ty_1 {
      | ---------------------------------------------- similarly named struct `twai_general_config_t__bindgen_ty_1` defined here

error[E0560]: struct `esp_idf_sys::twai_timing_config_t` has no field named `__bindgen_anon_1`
   --> /home/s413/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-hal-0.45.2/src/can.rs:179:21
    |
179 |                     __bindgen_anon_1: twai_timing_config_t__bindgen_ty_1 { triple_sampling },
    |                     ^^^^^^^^^^^^^^^^ `esp_idf_sys::twai_timing_config_t` does not have this field
    |
    = note: available fields are: `clk_src`, `quanta_resolution_hz`, `prop_seg`, `ssp_offset`, `triple_sampling`

Some errors have detailed explanations: E0422, E0560.
For more information about an error, try `rustc --explain E0422`.
error: could not compile `esp-idf-hal` (lib) due to 2 previous errors
```

Vollbrecht, 2025-05-13:
did you test out a release version of esp-idf? esp-idf master is pretty unstable to test against, maybe is just something they are changing in the driver or one of many other things.

syx-413, 2025-05-15:
I will soon to test,

closed 2025-05-24 by syx-413

smallbraineng, 2025-08-09:
I am getting this error on the v5.5 release

ivmarkov, 2025-08-09:
> I am getting this error on the v5.5 release

5.5 is not yet supported. If you want to have it supported, best is to contribute a PR that fixes this.