esp-rs/esp-idf #528: RMT doesn't work on ESP32S3

weixiongmei, 2025-05-24:
## Bug description
```
I (22) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:07:32
I (24) boot: Multicore bootloader
I (28) boot: chip revision: v0.1
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (77) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (85) boot:  3 factory          factory app      00 00 00020000 00200000
I (92) boot:  4 ota_0            OTA app          00 10 00220000 00200000
I (100) boot:  5 ota_1            OTA app          00 11 00420000 00200000
I (107) boot: End of partition table
E (111) boot: ota data partition invalid, falling back to factory
I (118) esp_image: segment 0: paddr=00020020 vaddr=3c060020 size=253dch (152540) map
I (154) esp_image: segment 1: paddr=00045404 vaddr=3fc93100 size=0341ch ( 13340) load
I (157) esp_image: segment 2: paddr=00048828 vaddr=40374000 size=077f0h ( 30704) load
I (167) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=5830ch (361228) map
I (233) esp_image: segment 4: paddr=000a8334 vaddr=4037b7f0 size=078d0h ( 30928) load
I (247) boot: Loaded app from partition at offset 0x20000
I (247) boot: Disabling RNG early entropy source...
I (259) octal_psram: vendor id    : 0x0d (AP)
I (259) octal_psram: dev id       : 0x02 (generation 3)
I (259) octal_psram: density      : 0x03 (64 Mbit)
I (264) octal_psram: good-die     : 0x01 (Pass)
I (269) octal_psram: Latency      : 0x01 (Fixed)
I (274) octal_psram: VCC          : 0x01 (3V)
I (279) octal_psram: SRF          : 0x01 (Fast Refresh)
I (285) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (291) octal_psram: BurstLen     : 0x01 (32 Byte)
I (297) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (303) octal_psram: DriveStrength: 0x00 (1/1)
I (308) MSPI Timing: PSRAM timing tuning index: 5
I (313) esp_psram: Found 8MB PSRAM device
I (318) esp_psram: Speed: 80MHz
I (341) mmu_psram: Read only data copied and mapped to SPIRAM
I (382) mmu_psram: Instructions copied and mapped to SPIRAM
I (382) cpu_start: Multicore app
I (776) esp_psram: SPI SRAM memory test OK
I (784) cpu_start: Pro cpu start user code
I (785) cpu_start: cpu freq: 240000000 Hz
I (785) app_init: Application information:
I (787) app_init: Project name:     libespidf
I (793) app_init: App version:      1
I (797) app_init: Compile time:     May 24 2025 00:18:18
I (803) app_init: ELF file SHA256:  000000000...
I (808) app_init: ESP-IDF:          v5.3.3
I (813) efuse_init: Min chip rev:     v0.0
I (818) efuse_init: Max chip rev:     v0.99 
I (823) efuse_init: Chip rev:         v0.1
I (827) heap_init: Initializing. RAM available for dynamic allocation:
I (835) heap_init: At 3FC96F98 len 00052778 (329 KiB): RAM
I (841) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (847) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (853) heap_init: At 600FE000 len 00001FE8 (7 KiB): RTCRAM
I (859) esp_psram: Adding pool of 7616K of PSRAM memory to heap allocator
I (867) spi_flash: detected chip: gd
I (871) spi_flash: flash io: dio
W (875) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
W (888) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (899) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (908) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (918) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (927) sleep: Configure to isolate all GPIO pins in sleep state
I (934) sleep: Enable automatic switching of GPIO sleep configuration
I (941) main_task: Started on CPU0
I (945) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (954) main_task: Calling app_main()
E (958) rmt(legacy): rmt_set_gpio(553): RMT CHANNEL ERR
E (964) rmt(legacy): rmt_config(699): set gpio for RMT driver failed
Error: ESP_ERR_INVALID_ARG
I (973) main_task: Returned from app_main()
```

## To Reproduce
Run the RMT Example Code

Vollbrecht, 2025-05-24:
the esp32s3 has more tx/rx channels. 0-3 are tx channels and 4-7 are rx channels. Can you try to change the rx channel from 2 to 4?

weixiongmei, 2025-05-24:
> the esp32s3 has more tx/rx channels. 0-3 are tx channels and 4-7 are rx channels. Can you try to change the rx channel from 2 to 4?

Thanks, problem solved. Where can I find the infomations about which channel is TX which are RX? I Don't see it in the API Doc. Also, I forgot where I have saw the esp32 has total of 16 RMT channels, is that correct? Or there are only 8?

weixiongmei, 2025-05-24:
@Vollbrecht Ok, thank you. There's another issue I'm having now, it doesn't allow me to change the mem_block_num, any number other them 1 will give me the following error

```
E (963) rmt(legacy): rmt_internal_config(583): RMT MEM BLOCK NUM ERR
E (969) rmt(legacy): rmt_config(700): initialize RMT driver failed
```

closed 2025-10-20 by ivmarkov